BelfrySCAD/BOSL2 #1642: write() API

adrianVmariano, 2025-04-21:
`text` = can be a string or an array of strings.  The string can have newline characters, in which case it is treated as an array of strings

`width` = (maximum) width the text can occupy.  If no font size is given then the size is chosen to fit this width in one line.  If a font size is given the text is wrapped to fit in this width.

`height` = (maximum) height the text can occupy. If no font size is given, the text size is chosen to fit this height.  If a font size is given the text is wrapped to produce the shortest width constrained by this height

`box=[width,height]`, alternative parameter for passing width and height more compactly. 

If width and height are given together with no font size then the text is fit into the specified box with wrapping at the largest possible font size.   If font size is given then the text is 

`font_size` = Size of the font in conventional font size, adjusted by the 0.72 factor.  (So `0.72*font_size` gets passed to OpenSCAD.)

`Msize` (`Mheight`?) = The font size is chosen so an upper case M has the specified size.  

`align`= Set this to "left", "right", "center" or "justify".   Note that prototype uses "justify=" but it seems that proper typographic terminology (and the way I've always heard it) was that "justify" means simultaneous right and left alignment.  (practicaltypography.com says "justified text is spaced so the left and right sides...both  have a clean edge.  The alternative...is left-aligned text.")   I also noticed that the book says not to use justification without hyphenation.   Should `align` also accept  LEFT, RIGHT and CENTER?  

`align_last`= alignment of last line in multiline text.  Defaults same as align except when align is "justify" in which case defaults to "left".  If there is only one line then this method applies.  

`valign`= "top", "center", "bottom".    Vertical alignment to use when text doesn't fully fill a box in the vertical direction.  Default: center.  (As above, should TOP, CENTER, BOT, BACK, FWD work?)

`vcenter_mode` = control how Roman text, at least, is centered vertically.  As noted by Jordan, at least for small texts with 1-2 lines, it matters if the actual text has ascenders or descenders, so there are different centering choices here.  I'm not sure what exactly the modes should be.  Actually this concept also connects to the question of margins and indeed how much room you think there is to fit the text, so maybe the name is wrong. That is, does the text end at its actual top or nominal top, and at the actual bottom?  Or descender bottom?   It seems like some possible choices for how to define the text extent are:  nominal text box dimension, actual text size, actual text size ignoring descenders

`letter_space`= gives a space to add between letters. practicaltypography.com suggests that this be given in relative units, like 5%-12% of the font size, when applying to ALL CAPS.  That would suggest accepting a fraction or percentage value.   On the other hand, if the goal is to spread letters for 3d printability, maybe an absolute letter space is better?  This has to be handled manually in the code by breaking up letters and inserting the space due to the broken letter spacing method used in OpenSCAD.  If letterspacing is zero, do NOT break up words and insert zero, instead feed whole words to ensure that ligatures and kerning work optimally.   Of course, letterspacing can be a negative value to squish letters together.  

`line_spacing` = a multiplier that controls line spacing.  

`margin`=  (is this needed) specifies a margin based on the text somehow.  The notes say (interline-ascender)/2.  The point here is that an absolute margin is easily created by just changing the box size, so this parameter is needed only if it does something relative to the text.  

1. Wrapping of text will apply only to horizontal text.  Text written in the vertical direction does not get wrapped.  
2. If text is an array (or contains explicit newlines) then no automatic wrapping will be done.  (Including a length1 array.)
3. If text is at a specified size and doesn't fit in its width/height constraint then that's an error, ideally with a note on how much the text needs to shrink.  
4. Jordan noticed an issue about spacing and balance having to do with the ascenders and descenders in texts which I tried to capture with the vcenter_mode concept above, but maybe it should be something about vboundary---how the boundary of text is defined vertically?  I'm not sure.  I've also somewhat forgotten the details of the examples, but it was really applicable to things with only 1-2 lines of text and probably not relevant to long texts.   There's an example in the wiki for text() where "Foobar" is centered vertically and it looks wrong, presumably because it's centering on the abstract box which has a bunch of extra space above the text.  

For anchoring the obvious simple thing is to anchor on the bounding box of the text.  There is one point of uncertainty, which is, so you anchor on the actual bounding box of the realized text, or on the nominal requested bounding box.   I guess there could be an anchor mode for that.

There is a second question about anchoring to the baseline.  This is complicated because (1) there are multiple baselines in multiline text and (2) after saying "baseline" you still have to specify if you mean left, right, or center on the baseline, and for write3d you'd have 6 more directions to manage.   My best idea for how to manage this is that we have a function like LINE(dir,n=0) that gives you the named anchor text string for a particular line (or the first line by default) located at the baseline in the direction given by dir.   So LINE would produce some text that might look like `"baseline_[1,0,0]_4"` for `LINE(RIGHT,4)` and named anchors could be created for all those text strings.  This would be easy to do if we had an array of coordinate values providing the baseline locations for each line and the left, right and center for each line (which would depend on whether we're doing actual text extent or bounding box).

amatulic, 2025-04-21:
Approaches to letter spacing, for example `letter_spacing=1.2`:
1. Compute a space to add to each character equal to 20% of the width of uppercase N.
2. Compute a space to add to each character equal to 20% of the width of the normal-spaced text.
3. Compute a space to add to each character equal to 20% of the font size.
4.  Add 1.2 units to the spacing of each character.

My preference is option 4. The first two add an element of unpredictability from the user's perspective. I don't like the third one because it uses font height to control horizontal spacing, ignoring the aspect ratio designed into the characters of the font (i.e. narrow vs wide).

You know exactly what you're going to get with option 4 In that case the default letter_spacing would be zero, and negative numbers would compress the spacing.

adrianVmariano, 2025-04-21:
I would argue that options 1 and 2 are wrong.   Options 1-3 are all relative-to-the-font options, and the standard way to be relative to the font is to work relative to the em, the font box size.   Option 1 is going to be annoying for Hebrew...which even if the font has an N it may not be a meaningful referent.  Option 2 sounds like it's relative to the text size which is catastrophic, reprising the current OpenSCAD bug.  

As far as I can tell, the industry standard for letter spacing appears to be to compute a space to add as a percentage of font size.   The typography book I found gives 3 examples of setting letterspacing in different software and they are all (including CSS) shown as a percentage of font size.  This is predictable from the user's perspective.  I'm not sure what you mean about ignoring aspect ratio.  It is of course the case that you will choose different letterspacing for different fonts.  The typography book does suggest that 5%-12% is correct for SMALL CAPS, which note is a range, not a single value.  

I would say that the industry standard of relative spacing is *clearly* the right choice for general typography because it means you can make small font size adjustments and your letter spacing adjusts with it, e.g. if you switch your font size from 4 to 5.  So if the goal is to produce good looking output, the relative measure makes more sense.   If you're trying to letterspace ALL CAPS text because that's the typographically recommended thing to do you'll prefer having relative spacing.  

The case for a constant option is that you are adding space not to make the font look good but to space the letters far enough apart for printability.  A downside is that if you don't know the font size (because it's being selected automatically to make the text fit) then things will tend to go awry if the anything changes.  If you change the bounding box, or change the text, you might need to make a big change to the letter spacing.  At least if the goal is good looking text.  You don't have access to the size of the font when it has been chosen automatically.    I suppose the flipside is that if you **really** want 1mm between letters for printing needs then you also don't know how to get that if the font size is adjusting all over the place.   This would be the argument for specifying letterspacing in two ways, relative or absolute.  

One other observation: letter spacing should only be inserted in between letters in a word and not at the end of a word.  Maybe this is obvious, but I saw an article about how web browsers do it wrong, so apparently not obvious enough.

adrianVmariano, 2025-04-21:
Had an idea for dual units on letterspacing, that might be clever or maybe it's a kludge.  Define EM=[1] and the user can say `letterspace=.05*EM` to get relative spacing or `letterspace=1` to get absolute spacing, with the cases distinguishable because one is a number and one is a list.

amatulic, 2025-07-31:
For letter space units, I recommend we go with what the CSS standard has been using for years as a 'standard character' unit: `ch`. That is, `1ch` is the width of a "0" (zero). The zero seems to be the most "representative" glyph of a font set in terms of width. Another unit is the `em` but the width of an M character can vary widely with respect to other characters depending on the font.

I am a bit concerned about what we do with kerning when the letter spacing is something other than 1.0. It wouldn't be hard to generate a kerning table using textmetrics, and use that data to derive the 'advance' value of the first character of a pair, which would then be used to adjust letter spacing. This would either need to be done for every invocation of write() for every 2-character combination that appears in the text string, or we'd provide a separate function to generate a complete kerning table for a given font face, and the user passes that table into write() as an optional parameter. I suspect the first way, generating a small table for each text string, might be more efficient than generating one large table with thousands of combinations that won't be used.

adrianVmariano, 2025-07-31:
Based on my font research that I posted earlier, if relative letterspacing is to be specified, it should be done in the customary fashion relative to the font size (i.e. the em, which is not the size of an M but the size of the font design box).   The em is the font's basic yardstick and is the normal way to create measurements relative to the font's size.  Trying to measure specific glyphs is a hack and creates complexity and makes everything less robust to non-Roman lettering.   

According to practical typography, you want 5% to 12% letterspacing with all caps lettering and to do that in CSS you should set the letter-spacing property to some value between 0.05em and 0.12em, so the recommended method in CSS is to use the "em" unit.  I am not sure why we would want to do anything different than this.  What is wrong with the standard practice?  And do you really think the size of the zero is indicative of anything meaningful about font size for Chinese?  

I do not understand your concern about kerning.  If you use textmetrics to get the layout for a row of text any kerning that is being done is already in there, isn't it?  If you use that as your starting advance data before adding letter spacing then the kerning is automatically integrated.   Were you thinking of calling textmetrics a single letter at a time?  That's the only way kerning becomes an issue, but why would you do that?  

To me it seems like ligatures are the complication.  Whether they are needless or not is unclear.  If you spread ff apart in a font that is supposed to have an ff ligature it may look kind of funny.  But it will also look funny if you typeset it as a single glyph with added letter spacing.  It seems like letterspacing a ligature either requires a new "looser" ligature (for positive spacing) or it requires non-ligature glyphs with reduced space to produce something that resembles a looser ligature.  That's a terrible mess, so regardless of how desirable it might be to make ligatures work, it's impossible.  

Hmmm.  I guess ligatures are a reason why kerning could be problematic.  Now I'm wondering...does OpenSCAD even produce ligatures?  What happens if you run textmetrics on a text with ligatures.  Like if you feed it affit and there's an ffi ligature, does it give you a 3 glpyh return?  If it works that way then unless there's a way to specify a "no ligatures" option to a font it's going to be impossible to get a kerning between two f's in a font that has an "ff" ligature.  

Note that it seems absurd to try to make a complete kerning table.  Unicode is big, and if you square it....you're out of space.   And what would the point be when you just need the pairs that appear in the text?

adrianVmariano, 2025-07-31:
Ok, so I forgot how textmetrics works, so some of what I said above is wrong.  It seems like unless there's a way to disable ligatures, doing kerning correctly is impossible because you can't tell whether a reduced advance for a letter pair is due to kerning or the use of a ligature.  I guess you can assume it's always kerning and hope for the best.  

Who cares if there is an M in the font???  That doesn't enter into this.  The typographic standard font reference yardstick is the em and that's what we should be using for relative font spacing.  Anything else just invites confusion.  Yes, it's true that this doesn't have a universal clear and obvious connection to the font glyphs.  But it is still the standard yardstick, just like it is the standard method of specifying a font's "size".    Suppose someone tries to follow the typographic advice to add 5% - 12% letterspacing for all caps.  They CAN'T DO IT if you don't provide a way to letterspace relative to the em.  It sounds sort of like you want to redefine font standards to make them "more intuitive".   That way...we can have multiple conflicting approaches to dealing with fonts!  What a great simplification that would be!  Especially if everyone did it....with their own ideas of what makes things more intuitive.  

I confirmed that openscad uses ligatures (with Calibri font) having both ff, fi and ffi ligatures.  Things don't seem to go horribly awry if you pretend ligature dimensions are kerning, so it might work ok at least sometimes to make that assumption.

amatulic, 2025-07-31:
I'm not redefining a standard, I'm proposing to use an *existing*, more modern and intuitive standard that was developed for digital web fonts. But OK, I can hold my nose and go with em. What is the em size, if not related to the letter M? I know the size of the zero, height and width.

One size that would be more useful than `size` is an existing standard as ancient as moveable metal type: cap height. This would correspond to nominal height (excluding ascenders and descenders), and would be exact; the write() module would just scale the font size so the nominal height equals exactly what the user specifies. The `cap_height` parameter would be mutually exclusive with `font_size`. When I am 3D printing fonts, especially numbers and labels that are all caps because lowercase doesn't print well, all I am interested in is cap height, and it takes some experimentation with size to get it right.

amatulic, 2025-07-31:
Cap height comes from here: https://en.wikipedia.org/wiki/Cap_height
Yeah, Msize is the same thing. Cap height seems to be the standard term. In any case it would be mutually exclusive with font_size.

My memory could be wrong, but I recall this corresponded to the nominal height in fontmetrics.

adrianVmariano, 2025-07-31:
[image]

There's the nominal.ascent box in blue with some capital letters.

adrianVmariano, 2025-07-31:
I believe this is why we thought we'd use Msize instead.  Or Mheight.   Considerably effort went into the API that we have already worked out, so if you think you need to deviate from it we should figure out if there's a reason it ended up the way it did.

adrianVmariano, 2025-07-31:
Depends on what you mean by "where it came from".  The 72% problem is a bug plain and simple where an extra factor of 72 is in the code (because 72 points was around for some reason---maybe it was a default dpi for the output device?  It doesn't really matter).   The reason it appeared plausible is that the cap height is typically around 70% of the em.  

Textmetrics obviously returns a different value for the height of a capital M compared to the return from fontmetrics as seen by my image above where they obviously aren't the same.

amatulic, 2025-08-01:
Yes, in fontmetrics there's a nominal ascent and a max ascent. We had found that the max ascent is for the diacritical marks, and the nominal ascent contains the ascent for normal characters. The more useful values come from textmetrics, however.

I can use the textmetrics ascent property for the M and it's always exactly the height of the M. I can then use that to scale the font size to get exactly the height I want.

For example, Garamond size 10 M in OpenSCAD has an ascent of 8.81, so it needs to be scaled up by 114%. To get a M height of 20, you need a font size of 22.88. Here it is overlaid on a 20x20 square.
[image]
In the other direction, Jokerman size 10 M in OpenSCAD has an ascent of 11.68, requiring scaling down by 85.6%. To get a cap height of 20, you need a font size of 17.1. In this case, however, the M has a descender, but after some experimenting with other decorative and calligraphy fonts, I think anything below the baseline can be ignored for getting a desired cap height or M height.
[image]

adrianVmariano, 2025-08-01:
The point here is that the "cap height" is a property of the font, not a property of a glyph.  If such a property exists, we do not have access to it.  Or at least, not with the behavior we want.  The nominal ascent is taller than capital letters.  And the max ascent is even taller than that.   My picture above shows NOMINAL ascent and there's a lot of space above the ABC.   I think the characters that reach the max height are things like the line drawing unicode characters.  

[image]

There are some A's where the nominal height is green and the max height is red.  

We can compute the properties of individual glpyhs, but we don't know how that generalizes.  So maybe the font has a weirdly small M for some reason.   You can't declare that the size of the M is the "cap height".  It's just the size of the M.

adrianVmariano, 2025-08-01:
It seems silly to support paragraphs, but OK.    That seems to create a variety of complications and may mean that things about newlines in the API are wrong.   And it seems like it supports a rare edge case while making more useful things harder---small texts are vastly more common in 3d printing than multi-paragraph. 

First, if there are explicit newlines in the text what happens?  They will always be treated as paragraph breaks?  This means of course we need more parameters, a paragraph indent parameter and a parameter for the space between two paragraphs.   What happens if I provide a width only, no font size, and a text with one newline (hence two "paragraphs")?   I understood that in this case, the font size is picked so that the text fits in the width with the two rows of text separated by the newline.   But if newlines mark paragraphs and wrapping is always allowed, I don't know what this does.   

What if I want 5 lines centered and sized to fit a width.  Can I do that somehow with the right setting of paragraph parameters and the use of newlines?  Or I would need to pass an array of 5 lines of text?   I guess I'm making an assumption that you consider entries in an array lines and not paragraphs.  If I pass an array of text strings, what happens?   Are the entries in that array lines or paragraphs?  

I'm not sure if your fonts are different or if you're just being lazy and not testing claims before you make them.  My test reveals that even with more diacriticals, the letter stays in the green region defined by nominal ascent.  

[image]

Be aware that functional anchoring is a crucial part of BOSL2 objects, and not something that can be swept under the rug.  I think the basic issues of anchoring to a bounding box are clear?  You know what to do there?  The main question is the whole baseline business.  

The only risk I see about not thinking about baseline or line anchoring now is if it turns out to be annoying to figure go back and extract the relevant reference points later.   Ultimately I think we need anchoring to every baseline, so you need to have access to all the baselines.  As long as that's available, it's going to be straight forward to do whatever anchoring we need.  The anchoring problem is trivial to code.   The hard part is the design, not the implementation.

adrianVmariano, 2025-08-01:
I'm aware that sometimes paragraphs are separated by an extra space without indentation---not a "blank line" but a paragraph separation space.  And while yes, many paragraphs aren't indented, more of them are indented, so if paragraphs are supported, indentation seems necessary.  You *are* doing typesetting, especially if you support paragraphs.  That's why I thought supporting paragraphs was a weird idea.   But really any time you're creating text you're typesetting.  

If there is a default text size, how do you get it to pick the text size to fit the region---that's one of the most useful features I think.  The API does NOT have a default text size (and I actually don't like the idea of a default text size---why provide behavior that's always wrong?).    I think the most common use case by far is going to be putting short pieces of text into regions of specified height, width or both where you want the text to scale to fit the region.  

Do I have this right?  if I give a short line of text and a large width the text appears at some random default size.  And if I then disable word wrapping that causes the text to expand to fill the width? 

So newlines and array of text are internally equivalent.  I think that's good.   Treating lines as paragraphs, though, seems unfortunate.  You think you can get away without paragraph controls, but you really can't, because another very important case is going to be a user with a few lines of text like 2-3 lines that they want to set without wrapping.  They're going to want normal line spacing but they want to determine where the newlines are.  Because these are treated as "paragraphs" there needs to be a way to set the interparagraph space equal to a normal line leading space.   So I don't see a way to avoid having an adjustment for the space between paragraphs.   I would think that the default needs to be no wrapping and no extra space between paragraphs.

adrianVmariano, 2025-08-03:
Wrapping wide text is not the same thing as supporting paragraphs.  It sounds from your description like you have simply implemented wrapping.  I would suggest that implementing full paragraph support can probably be avoided, then.   If you really want to you could provide some mechanism to indent the first line of the text.  And if a user wants spaces between paragraphs then they can do it manually.  

No, users cannot just scale things, because scale doesn't work with anchoring.  It will cause the object to be anchored incorrectly and then it will scale all the children, so you basically can't use scale() in BOSL2.   The one time I labeled things I **absolutely** would have used the ability to fit text to a box because I wanted the largest text size that would fit in the space to maximize readability.  I did this by trial and error with a size parameter instead.   Jordan's label example I think also involved sizing the text to fit the space.  I think this is actually doing to be the primary, dominant use case.   The idea that "the size is knowable" is missing the point.  The user doesn't want to have to do a bunch of shenanigans of creating text, calculating its size and then rescaling it (even in cases where this doesn't create problems due to anchoring).  Why make this common task so complicated?  The whole point of providing this module is to make these common tasks simple!  Also, keep in mind that fitting text to a box means potentially wrapping the text differently.   

Is there something wrong with how I wrote the API that is at the top of this issue?   I **thought** I had laid out my ideas clearly, and you seem completely unaware of them at basically every step, so I'm wondering how I failed to communicate here.  

The question about how to set text height from text relates to this:  why is a user doing this and what result is expected?  If the goal is to create text of a known (total) height then descenders should be included.   That seems like a natural goal.  Are there other reasons to want to do this?  What other goals might there be, where we can justify ignoring descent?  If you get too far away from returning the actual full height of that it's not functionally different than using an arbitrary font yardstick like the em whose relationship to the font is arbitrary...and then what's the point?  

I shouldn't have said "random" default size.  The problem with a default size is that it's **arbitrary**.  Why should there be an arbitrary default?  There's basically no reason to believe anybody ever needs text of that size.  Why 10 and not 8 or 12 or 15, or 100?   Do you routinely make text of size 10? 

In this case, there's elegance in the API created by allowing the font size to be left unspecified as an indication that the module is supposed to choose the size and hence not wrap.  The alternative is much clumsier, where you have a size that sometimes gets ignored.  And you need an extra option to basically say "choose the font size".  And you have to confirm that the user didn't specify a size with some other option because they are inconsistent.  

Top is your modified Lamda font.  Bottom is the NanumGothic ExtraBold I used for 3d printing.  These are at the same glyph height (which included the descender portion of my font):

[image]

My font has thicker lines, which is better, but it's wider, which is a downside.

amatulic, 2025-08-05:
OK, now we're getting somewhere. Your analysis helped clear up what was disturbing me about the API which I couldn't adequately express. How about this, mostly the same as what you wrote:

| Font size | width | height | Response |
|------|------|------|------|
| known | undef | undef | Works just like text() with arrays/newlines supported |
| known | finite | undef | Text wrapped if required to fit in width |
| known | undef | finite | Meaningless, assert error |
| known | finite | finite | Text wrapped to fit in width, echo warning if height exceeded |
| unknown | undef | undef | Meaningless, assert error |
| unknown | finite | undef | Size text to fit width, no wrapping |
| unknown | undef | finite | Size text to fit height, no wrapping |
| unknown | finite | finite | Size text to fit either constraint without exceeding the other, no wrapping |

The above conditions mean:
* a font size is required to wrap text, which is what I have been advocating all along.
* font size must be undefined for auto-scaling, in which case there is no wrapping.

For that last line in the table (your "unknown font case 4") I prefer to avoid the complication of finding a font size that results in wrapping within the defined box, I'd rather just not wrap it at all.

amatulic, 2025-08-05:
"Take this line of text and wrap it to 3 rows" is not the same thing as "take this line of text and wrap it to fill but not exceed 50 mm of height."

Thinking about it more, it could be done with my existing wordwrap optimization, if "height" equated to "total number of lines after wrapping". I suppose that would be `floor(box_height/line_height)`. The wordwrap optimization iteratively finds the minimum width that doesn't result in increasing the number of wrapped lines. If you want N lines, then N is the constraint and it would find the minimum width that results in N lines.

If I want to know if my wrapped text fits inside a box, I'd like a warning that tells me when it doesn't. Specifying a height and width with a font size is perfectly valid if the wrapped text fits inside the specified box.

amatulic, 2025-08-05:
Wrapped text that fits inside a box meets the constraints. Would there be a case for using anchors with that box? If the text goes beyond the boundaries, it's trivial to echo a warning message, although it's probably obvious simply by viewing the rendered result.

There are three bounding boxes possible with these combinations of height and width bounded or unbounded:
1. the box bounding the final text, which is always narrower than the specified width
2. the container box initially specified, if both dimensions are bounded
3. a hybrid box consisting of the specified container bounded dimension and the size of the final text bounding box

In the prototype, box 1 could be positioned within the container width, left, right, or center. Here, if you specify a full box to contain the text, one has the ability to position the final text bounding box at the top, center, or bottom of the container box. Emitting a warning if the height of the container is exceeded isn't necessary but I thought it might be useful to have.

adrianVmariano, 2025-08-20:
Actually we should be saying something like "anchor box" to distinguish from other boxes that might be flying around.  It occurs to me that the remark from Jordan about font sizing could be taken to suggest that you want an anchor box sized to include ascenders, but what he really means is that you fit the text into the bounding box in a way that respects ascenders---even when none were used.

adrianVmariano, 2025-08-21:
No, I am insisting on font size by specifying the em, which has nothing to do with the letter M.  Totally different things.  The "Msize" was meant to be the height of the letter M.   My idea is that `size=` would be the OpenSCAD size for compatibility and then `font_size=` would be the conventional font size (scaled by 0.72).   Calling the latter them em_size would be technically correct, but confusing to users who have been specifying "font size" all their lives in every program.    I have no objection to calling the openscad size osize, but that is a bit more of a speedbump.  I don't know if the added clarity is helpful or not.  
Probably should be `cap_height` for consistency.  Not sure if `lheight` is ok or if it should be `line_height`.  

It's totally reasonable that you have a function to get the font size from all of these possibilities and also to issue error messages if the user has given two of them at once.  And also I think appropriate to expose that function to users.   (But document it last, after other text functions that are more directly useful.  There's a natural tendency to write the supporting functions above and the meaty functions later that we have to override to put things in the right order for the docs.)  

I don't think we should overthink the sizing.  Whatever information comes out of textmetrics is what we have to work with.  I'm not sure I understand the issue about a possible need for refactoring.  

So the proposed anchor boxes were:

1. user specified bounding box
2. tight box (as much as possible) based on glyph sizes
3. box sized to accomodate ascender and descender size
4. box sized based on line spacing

Is that right?  Are you planning on doing 1, 2 and 3 but not 4?  Between 3 and 4 I'm not sure which is more useful.

amatulic, 2025-08-21:
When I wrote "M height" I meant "em height", and I did say that for this I'm using the distance between the nominal ascender and nominal descender. I intend to rename lheight to line_height, mheight to em_height, and capheight to cap_height for consistency with BOSL2's existing conventions. I also have osize (OpenSCAD font size), and I'm not sure what to call it. Calling it o_size makes it seem like it represents the size of a lowercase o.

Refactoring the wordwrapping, which relies on individual character positions calculated from kern-corrected advance properties, *might* be a big job if I were to account for where the right side of the last glyph is supposed to be, and I suspect it would not be reliable for all fonts. I need to do some experiments. Do some fonts have empty space to the left of the glyph for the letter H, and if so, how would I get the space on the right, which is what I really need? How are glyph width and advance affected by italics? At the moment it's easiest simply to use `advance` and count on the fact that the advance position and right edge of a glyph don't bracket the box boundary. Given that this difference is typically a tiny fraction of a unit, this would be a rare occurrence so I'm not sure if refactoring is worth the effort.

Because I'm using the new object() feature, I can maintain an object for all box sizes.
Boxes 2, 3, and 4 would differ only in height. Box 2 and 3 would already include whatever line spacing is required or requested, so I am not sure if box 4 is needed but it can be available.

adrianVmariano, 2025-08-21:
> When I wrote "M height" I meant "em height", and I did say that for this I'm using the distance between the nominal ascender and nominal descender. This is different from cap height, which is the actual vertical size of the letter H glyph (or whatever $refchar is set to).

Not sure what you mean about distance between nominal ascender and descender.  I think that if the user requests a font_size of x the code should pass 0.72*x as the openscad size to the font engine.  This doesn't depend on any font metric information.  I also think it should be called "font size".  Note that the em is not a height so calling it "em_height" is wrong.  The em is a length.  But the standard font terminology is that when you specify a "font size" you are giving the size of the em, which is why "font size" is the correct thing here.  Calling it "em_size" would be accurate but I think is an obfuscation for the regular user.   It is perhaps worth running the names by the folks in the chat.  I agree that o_size could be confusing, which leads back to having `size=` for the openscad size, `font_size=` for the standard font size.  

I wouldn't worry about tiny errors in the anchor boxing.  I can't think of a situation where somebody needs to have the font end exactly at the box edge.  Users would be doing things like centering the box somewhere, or placing the box inset a few units from the edge, in which case a tiny error will not be noticed.

amatulic, 2025-08-21:
Historically, the em was the height of the metal type, which includes the ascender and descender. In the past we found a few fonts that don't follow the 72% relationship, so I prefer to derive the em from metrics if possible. This would make any font appear consistent in size next to any other font of the same em size, and should still work out the same as the 0.72 relationship for most fonts. Doing it this way also avoids the need to make corrections if OpenSCAD's internal font sizing ever changes.

amatulic, 2025-08-22:
> You cannot derive the em from metrics. That is impossible. The em is the size of the design font box for the font.

For digital fonts, why wouldn't the em size simply be the maximum ascender-descender span, from the top of the box to the bottom? We have that information from fontmetrics. Presumably the maximum values define the total height of the design box. Why waste the extra space? Using this metric would also result in consistency when displaying two different fonts of the same size next to each other, more than simply multiplying the OpenSCAD size by 0.72.

While I think `cap_height` would be the most useful size (at least for me), the second most useful size would be `nom_height`, which is the span between the nominal ascender and descender. This would allow a user to specify the height occupied by all the usual uppercase and lowercase glyphs.

I can also make a `max_size` that uses the maximum ascender and descender, which would include all diacritics. And an `em_size`, which is divided by 0.72 to convert it to an OpenSCAD font size. It would be interesting to compare the two. Personally I am skeptical that either of them will be useful, though. Same for specifying size by interline height.

adrianVmariano, 2025-08-22:
Another observation based on my exploration of font sizing is that you're delusional if you think that using ascender/descender span will give you font size consistency.  The fonts seem to different considerably on the relationship between those values as reported by font metrics and the observed glyph spans.  Some fonts seem to be mysteriously small relative to their max ascent and some seem to bump right up to it.

amatulic, 2025-08-22:
I don't find this to be "such an issue". I'm just creating a variety of ways to specify font size. Now that you have explained em size, I multiply `em_size` by 100/72 to convert it to OpenSCAD font size. It's already done. I now have this:
````
// Function: get_fontsize()
// Synopsis: Return an OpenSCAD font size corresponding to an alternative size metric.
// Topics: Text
// See Also: write(), textwrap(), textobj()
// Usage:
//   size = get_fontsize(font, [cap_height=], [nom_height=], [max_height=], [iline_height=], [em_size=]);
// Description:
//   Returns the equivalent OpenSCAD font size corresponding to the specified alternative size metric.
//   The size parameters are mutually exclusive; one and only one of them must be specified.
//   .
//   The `cap_height` parameter is a useful size specification for uppercase labeling, and results in capital
//   letters approximately equal to what OpenSCAD's `size` parameter in `text` gives. The special variable
//   `$refchar` is used as the reference character for a capital letter, and defaults to `H`. If your text
//   uses only numbers, then you may want to pass `$refchar=0` in the argument list.
//   .
//   The `nom_height` parameter specifies the nominal height of common characters (e.g. A-Z, a-z, 0-9, punctuation)
//   including ascenders and descenders. This is useful for creating labels with mixed-case text.
//   .
//   The `max_height` parameter specifies the maximum height possible in the font. This would include characters
//   with double diacritics and drawing characters like vertical bars.
//   .
//   The `iline_height` parameter lets you specify a font size in terms of the font's internal interline height.
//   .
//   The `em_size` parameter is the standard unit size in typography, the size of the design box for the font glyphs.
//   The OpenSCAD font system has a bug: Specifying `size=s`, where `s` is the em size, results in text having
//   a size of `s/0.72`. For many fonts, this causes the size of capital letters (not the ascender-descender range)
//   to be approximately equal to `s`, because fonts commonly use about 70% of their height for their
//   ascenders. The `em_size` parameter here gives you the customary font size, converting to the OpenSCAD size
//   by multiplying `em_size` by 100/72.
// Arguments:
//   font = The font to use. Default: "Liberation Sans:style=Bold"
//   ---
//   cap_height = Height of a capital letter, using `$refchar` as the reference character.
//   nom_height = Height of normal characters, from nominal ascender to nominal descender.
//   max_height = Maximum height possible in the font, from maximum ascender to maximum descender.
//   iline_height = Interline height for the specified font; the resulting glyph size may be much smaller.
//   em_size = Standarf font unit size, the size of the em-box in which the font was designed.
function get_fontsize(font="Liberation Sans:style=Bold", cap_height, nom_height, max_height, iline_height, em_size, ) =
    assert(version_num() >= 20210816, "\nRequires OpenSCAD release after 2021-08-16.")
    assert(num_defined([cap_height, nom_height, max_height, iline_height, em_size])==1, "\nOnly one of cap_height, nom_height, max_height, iline_height, or em_size must be specified.")
    let( // arbitrarily use a size 10 font to calculate scaling factor
        fm10 = fontmetrics(10, font),
        tmc = textmetrics($refchar, 10, font), // $refchar defaults to H
        fontscale = is_def(osize) ? 1
            : is_def(cap_height) ? cap_height / tmc.size[1]
            : is_def(nom_height) ? nom_height / (fm10.nominal.ascent-fm10.nominal.descent)
            : is_def(max_height) ? max_height / (fm10.max.ascent-fm10.max.descent)
            : is_def(iline_height) ? iline_height / fm10.interline
            : 0.1 / 0.72
     ) 10*fontscale;
````
Wordwrapping now seems to be working as it did in the prototype version but it handles kerning and letter spacing correctly. Now I start introducing complications like paragraph indent/outdent and full justification.

adrianVmariano, 2025-08-22:
Don't you need a size= option for plain openscad size?  I favor font_size over em_size because the former is a standard term and the latter a jargon technical term.  Maybe we should ask others.   I forwarded to Jordan for comment, but he's traveling and I haven't heard anything back yet.  

Also do you need the option to return undef?   As written I would say "Exactly one of....must be specified".

amatulic, 2025-08-24:
> Don't you need a size= option for plain openscad size?

For the function I pasted above, it *returns* an OpenSCAD font size equivalent to one of the others. It seemed odd to pass the same parameter that would be returned. While this is function is exposed, it isn't used in any examples and any internal use first tests whether the font size is already an OpenSCAD size before calling it.

> I favor font_size over em_size because the former is a standard term and the latter a jargon technical term. Maybe we should ask others. I forwarded to Jordan for comment, but he's traveling and I haven't heard anything back yet.

The standard font size is now `font_size` throughout, with `em_size` used only in documentation to explain it.

> Also do you need the option to return undef?

It asserts an error if you don't specify one. One can't do anything with text without the font size, so undef wouldn't be helpful.

> Jordan thinks that having font_size and size both would be confusing and favors em_size as being more specific. He didn't have any other remarks on the proposal.

I feel that `size` is too generic. Throughout the code I consistently call it `osize` for "OpenSCAD size" to distinguish it from the other specific sizes supported. The function above is also now called `get_font_osize()`. Changing `font_size` back to `em_size` is just one find&replace operation. I see Jordan's point; `em_size` means a specific thing, while `font_size` could be interpreted to mean some other metric like OpenSCAD size, rather than em size.

adrianVmariano, 2025-08-24:
Well, it's your code.  But the way I would do it is have one function that computes the size and the size is unconditionally obtained from that function.  That means if openscad size is given, it's an arg to get_font_size() and that function basically operates as a pass-through.  This is cleaner and more elegant than separately checking if "size" is set, and 7 other things are not set (basically repeating error checking code) and then setting size and otherwise calling the font size function.   It also means that if you add a new font size method you have to change code in multiple places instead of just one.   Right now we have code like get_radius() which acts as a simple pass-through if the user happened to supply `r` as input---nothing wrong with that.  

It seems like if OpenSCAD font size is going to be an allowed user input it should be "size" to match the parameters used by every other text creating module.  Yes, "size" is unfortunately generic, but introducing a new name that means the same thing in a text context that "size" means everywhere else seems needlessly confusing.   Someone may use your sizing function with path_text, which uses size as its parameter, so they would be writing `size=get_font_osize(...)` where surely `size=get_font_size(....)` would be more natural and easier to remember.  

I thought that when you didn't have a font size it was going to fit the font to the box and disable wrapping.   That means it's legal to give no size and undef should be an option.   See your post from 3 weeks ago (Aug 5) with the table in it.  
 
Before you make the change let's poll Revar and Richard and see what they thing about em_size vs font_size.

adrianVmariano, 2025-08-24:
Richard supports em_size.  He also says "osize" sounds like you're talking about the size of an "O" so he opposes that term (like I do).

amatulic, 2025-08-24:
I didn't care for `osize` either for the same reason, but I left it that way because it's a unique name that is easy to replace globally. Once I set it to `size` there's no going back. I was considering something more descriptive too like `openscad_size`.

> I thought that when you didn't have a font size it was going to fit the font to the box and disable wrapping.

That's correct. That function to get a font size isn't called in that case, because there is no size set. That function is just for converting from one kind of size to OpenSCAD size.

I'll add OpenSCAD font size to it as a pass through, and change `font_size` back to `em_size`.

adrianVmariano, 2025-08-24:
How do you know that no size is set so you can avoid calling get_font_size?  You have to check all the sizes, which is a complication.   To me it seems simplest to structure the code as 

`size = get_font_size(size=size, em_size=em_size, etc, etc)`

and then you can check if size is undef to determine if a size was given.  Your approach is messier.  Again, it means checking that every size related variable is unset, so now you are enumerating all the size parameters in another place.   Why complicate things like that?   Again, it creates the problem that if you add a new size it has to be added in more places because you have a separate test that all the size parameters are unset.  Consolidate the whole thing into get_font_size() !

If you think it's important to be able to issue an error for an undefined size, have an `error=` flat to control that.   But I am not sure that's worth worrying about.

amatulic, 2025-08-24:
> Consolidate the whole thing into get_font_size() !

That was my original intent, but you advised against it, using the clunky example `circle(r=get_radius(d=12)`. So yes, `write()` and everything else exposed would have every kind of font size in the argument list. I'm actually making this simpler by making some functions private. Now that textwrap() returns an object instead of a simple text array, and it's now useful only to code that requires positions of each character, there's not much point exposing it anymore. I'm considering the same with textobject(), which returns a lot of useful information about a line of text (character positions, kerning, etc.). If I make that private, then the only thing left is write() and get_font_size() having the whole collection of sizes in the argument list.

adrianVmariano, 2025-08-24:
Huh?  Maybe you misunderstand.  All I meant with the clunky example is that the user doesn't normally call get_font_size, not that get_font_size does less.  

The way it should work is that get_font_size() completely handles the computation of the font size, whether by passing through a simple openscad value or computing something from the font.  But get_font_size is not called by the user under normal circumstances.  **ALL** possible font size parameters are passed to get_font_size, not just some of them like you were reporting.  In other words, it looks something like:

```
function get_font_size(font, size, Msize, em_size) = 
    let( num_parms = num_defined([size, Msize, em_size]))
    numparms==0 ? undef    // return undef if no size given
: assert(numparms==1, "Must define at most one of ....")  // error if conflicting size params
  is_def(size) ? size  // simple passthrough
: is_def(Msize) ? ....   // computed with font metrics
: em_size*0.72;     // simple computation

function write(font,size,Msize,em_size,...) =
   let( size = get_font_size(font, size=size, Msize=Msize, em_size=em_size))
   ... computation continues with size possibly undef if no size given ...
```

Maybe I'm misunderstanding, but from the code example and what you've written it sounds like you've split what I see as the natural function of get_font_size into two places and created unnecessary code duplication and complexity as a result.  

If somebody actually wants to use get_font_size in some other module they will have to copy your code out of write() instead of just calling get_font_size() because it has to be called wrapped with other checks.   In the future path_text() may want to call get_font_size, for example.  

With regards to your internal data structures, I would say that you should err on the side of making stuff private.  If we later discover that there's reason to expose it we can do it then, but keeping it private keeps the API simpler and means less work doing docs.   In the future path_text() may want to use your internal size table with kerning, but that can be handled as a private API rather than public since it's internal.

adrianVmariano, 2025-08-25:
As previously described, the get_font_size() function would need to be wrapped in checks for size/osize and possibly whether the size is defined at all.  If you've fixed that, that's good.  It seems reasonable to expose get_font_size.  

You **must** center things around the origin for anchoring to work, regardless of how the user has specified things.  The anchor box is going to be centered on the origin.  When a dimension is not specified (e.g. height=INF or width=INF) you will need to determine a height to feed to the anchoring system, which will be what determines where the text will actually appear when anchored.  Actually this part of dealing with the anchoring could be annoying.  Anchoring assumes a centered box, so you'll need to work with that.  This was one of the reasons I wanted a named "baseline" anchor.  

So if height is INF it seems like you could make the baseline anchor the default, maybe?  But the text probably needs to be centered.  (Though what exactly that means is not entirely obvious.)  

You might want to see what Revar and Richard think about these various default questions and text positioning questions as well.  The default anchor is usually CENTER, which would mean that the box holding the text would be centered.

amatulic, 2025-09-20:
Short two-letter value names are difficult to find reliably in the source code. I can change `em_size` to `em` but I'd do this as the last step before the PR is merged.

I've been working on this slowly. I'm about at the point where I have to figure out anchoring and attachments.
I imagine I'd have something like this:
````
attachable(anchor, spin, two_d=true, size=text_container_size) {
    display_text_centered_at_origin();
    children();
}
````
It can't be that simple. What am I missing?

adrianVmariano, 2025-09-20:
That looks right to me.  The complication will be named anchors to do baseline anchoring.

adrianVmariano, 2025-09-20:
You can change the user-facing parameter without changing the internal variable, so then it would only need to be changed in once place if we decide on an API change.

amatulic, 2025-09-20:
Amazing. It actually works.

Here are two paragraphs of size 10 text with indent=10, line_spacing=0.9, para_spacing=1.5, letterspacing=0.2, align="justify". Also shown are an inner box (`hfit="tight"` and `vfit="tight"`), an outer box (`hfit="width"` and `vfit="nominal"`), and the baseline of the top line (which would be the baseline if there was just one line). Full justification as well as anchoring uses the width of the box determined by `hfit`.
The anchor in this picture is `FRONT+LEFT` for the "tight" box, with `spin=45`.
[image]

> The complication will be named anchors to do baseline anchoring.

I don't have any idea how that's going to work. As shown in the picture, I do know where the baseline is for a single line within the bounding box, however.

amatulic, 2025-09-21:
I had envisioned this documentation:

>`atype` = Anchor type, when set to "box", uses the bounding box of the text, in which case `anchor` specifies the usual anchoring points (for example `FRONT+LEFT`). When `atype` is set to `BASELINE(n,on_text)`, the `anchor` parameter specifies the anchor location on the baseline of line `n` (starting at 0), with `on_text` (optional) specifying whether the baseline length is the physical length of the text (`true`, default) or the width of the bounding box (`false`).
>`anchor` = Translate so anchor point is at origin (0,0,0). If `atype` is a `BASELINE()` function, then only the horizontal components of `anchor` are used. Default: `CENTER`

...where `BASELINE()` is a function literal similar to what we were using in metaballs() to specify different kinds of metaballs in the argument list. I envisioned `BASELINE()` to return an object containing the anchor coordinates, which can be an arbitrary place inside the bounding box.

But then, I'd need a way to set an arbitrary point inside a rectangle to be the anchor.

adrianVmariano, 2025-09-21:
What if I want to anchor something to each line of a text?  Or the first and last lines of a text.  (Presumably anchoring to the right or left edge of the text or box, not the actual text itself, which is an unknowable mystery.)   That requires simultaneous access to multiple different baseline anchors for different text lines.   Remember that anchors are used both to place the object but also to place other objects relative to the object.  

There is an offset option in attachable that might sort of do what you were hoping but maybe also not really, and it might not work for trapezoid anchors, only prismoids.   Maybe positioning the text off-center could do what you want, but that might mess up other anchors (without the offset mechanism).  Don't forget that anchors aren't just positions, they have direction.

amatulic, 2025-09-21:
Ah. You mentioned the offset option. That does exactly what I want. I can position the anchor anywhere, and it still works with spin. So I could anchor to any baseline inside the box. I don't know what that does to stuff you'd anchor to the text, though. What I need is a test case.

amatulic, 2025-09-21:
Hmm. When I pass into attachable() anchor=LEFT and offset=[20,0], it does what I expect, offsets the anchor 20 units from the center left edge of the box. But when I pass anchor=CENTER, it doesn't do what I expect, it stays centered no matter how I set offset.

adrianVmariano, 2025-09-21:
It looks like you have to combine `cp=point` with `offset=-point` to get the desired behavior.

amatulic, 2026-09-01:
I've read through these comments and am still mystified about anchoring.

Here are the things I have.

* `tightboxsize` is the box of the tightest width that fits around the text, which may be multi-line. The height of the box includes all the lines accounting for ascenders and descenders reported in textmetrics() for the text, line height, and paragraph spacing.
* `nomboxsize` is a box of the width you specify to fit the text within. Width can be `INF` in which case no word-wrapping is done, but the input text can still be multi-line. Height is based on nominal ascenders and descenders reported by fontmetrics().
* `maxboxsize` is the same width as `nomboxsize` but the height accounts for the maximum ascender and descender reported by fontmetrics().

For baseline anchoring, box height is irrelevant; all we need are the widths.

There is also an array of baseline position objects, `baseline_pos[]`, and each element `i` has these properties:
* `baseline_pos[i].xstart` = x position of first character (left edge if LTR, right edge if RTL)
* `baseline_pos[i].xend` = x position of end of line (right edge of last char if LTR, left edge of leftmost char if RTL)
* `baseline_pos[i].y` = y position of baseline

All properties are relative to the origin, when `tighboxsize` is centered on the origin. The `xstart` and `xend` properties vary depending on text direction (ltr or rtl), paragraph indentation, and justification (left, right, center, justified) within the `tightbox` container.

I would like the default positioning of `tightbox` to be such that the first character of the baseline of the last line of text is on the origin. I imagine we would need anchor names like `anchor_text_0_left` for the left end of the baseline for line 0 or `anchor_text_last_center` for the center of the baseline for the last line. Similar names like `anchor_line_2_right` would set the anchor to the right edge of the box at the y value of the baseline for text line 2.

And then there's the attachments, for attaching more text. I might want to attach text with a different font or size to the end of line 3, for example.

Claude suggested this for the named anchor generator:
````
function _line_anchors(baseline_pos, width) =
    let(
        n = len(baseline_pos),
        boxw = (width == INF || is_undef(width)) ? undef : width
    )
    [
        for (i = [0:n-1]) each
            let(
                bp = baseline_pos[i],
                lo = min(bp.xstart, bp.xend),
                hi = max(bp.xstart, bp.xend),
                mid = (lo+hi)/2,
                // container edges: real box edge if bounded, else same as text edge
                box_lo = is_undef(boxw) ? lo : -boxw/2,
                box_hi = is_undef(boxw) ? hi :  boxw/2,
                labels = [str(i), if (i==0) "first", if (i==n-1) "last"]
            )
            for (label = labels) each [
                named_anchor(str("anchor_text_",label,"_left"),   [lo,     bp.y, 0], UP, 0),
                named_anchor(str("anchor_text_",label,"_right"),  [hi,     bp.y, 0], UP, 0),
                named_anchor(str("anchor_text_",label,"_center"), [mid,    bp.y, 0], UP, 0),
                named_anchor(str("anchor_line_",label,"_left"),   [box_lo, bp.y, 0], UP, 0),
                named_anchor(str("anchor_line_",label,"_right"),  [box_hi, bp.y, 0], UP, 0),
                named_anchor(str("anchor_line_",label,"_center"), [(box_lo+box_hi)/2, bp.y, 0], UP, 0),
            ]
    ];
````
For concatenating text, one would use:
````
write("Hello ", size=10)
    position("anchor_text_0_end")
        write("world", size=6, anchor="anchor_text_0_start");
````
Does that look OK?

amatulic, 2026-09-01:
Hmm. The example is failing because in `position()`, the internal `_find_anchor()` returns only the anchor name, not the anchor data.

amatulic, 2026-09-01:
Disregard. The problem was with Claude's use of `each` in the `_line_anchors()` function it proposed. Easy to miss.

adrianVmariano, 2026-09-01:
Ok, first I don't have time right now to read through everything in this long issue.  But I think the first thing to figure out is what are all the desired types of anchoring.  Once we have that list, we can figure out how to name them.  I'm hazy on the answer to the first question.  I also don't understand exactly what your desired default is or its motivation.  (My naive thinking is that the default should be centered in the bounding box at the origin---though as I recall we have different bounding boxes, which makes it complicated.)   Are there also different line ends?  

Things that come to mind:

* Anchor wrt bounding boxes of different kinds (all directions as usual)
* Anchor wrt left/right/centers of different lines (at baseline, top and bottom???)  Is a line start/end different than some other notion of its right and left sides?  (Or does that just flip for RL text?)

The first thing maybe should be achieved using normal LEFT/RIGHT etc anchoring with an atype to select between different boxes.  The second thing with named anchors.  It's dumb to have the word "anchor" in all the anchor texts, and I think we probably want helper functions that users use like `TEXT_LINE(4)` or `TEXT_LINE(4,<qualifier>)`

Have I missed anchor options?

amatulic, 2026-09-02:
I think you covered it, and I think so have I in my latest implementation. The difference is that I see no need for helper functions but I can certainly add them; the functions would simply return one of the anchor strings (and I've removed the word "anchor" from them). I want the default position to be the same as what OpenSCAD uses: the default is to position the left end of the baseline on the origin.

So far, everything seems to be working fine, but I wouldn't be surprised to find another weird bug like the recent thing with RTL text.

I have also implemented what we discussed in Gitter about the nonbreaking space, and I have made it handle kerning just like a normal space. A nonbreaking space is considered as a big box, like a letter H, and doesn't get any kern adjustments, but a regular space does if the prior character has a slope next to the space, like A, V, W. You can use `\u00a0` or `{ }` to make such a space. Spaces at the head or tail of the text are converted to nonbreaking spaces before any other processing is done.

I'll attach my write.scad when I'm happy with it, but I won't put it in a PR until I know that docsgen will work. The PR would involve changes to other files in the library to move text operations into its own file.

adrianVmariano, 2026-09-02:
One reason for helper functions is that TEXT_LINE(i) is easier to manage programmatically than for a user than `str("line_",i,"_center")` or something like that.  This is why I've introduced these for most (all?) places where I have indexed anchors.  

I think the default behavior should be that the text appears with its nominal box (the one the user requested) centered at the origin.  The whole point here is that we're creating text centered in a box, so the text should appear in a known and meaningful location relative to that box.  And in fact this is probably a large fraction of the use cases or more for how people will want the text positioned.   The other ones are that they will want it anchored on a box corner.  Positioned with the left end of baseline on the origin is a useless position that just means the default is never what users want. What OpenSCAD text() does is irrelevant: it doesn't produce text in a box.  

So what options do you have for anchoring to the overall box?  Do you have atypes?  Or some other scheme?  

If I'm understanding the above, you have something like line_##_<location> and text_##_<location> where "line" means the edge of the bounding box and "text" means the edge of the text.  The <location> can be either right, left or center.  So the only vertical option here is baseline?   It seems like someone might want an anchor located at the vertical center of line N of the text (for some notion of vertical center)---if they want to place some non text object adjacent to the text.   

Do you also have text_##_start and text_##_end, which would be different with RL text than LR text?  Or is the idea that you just choose the correct one (left/right) when joining text.