BelfrySCAD/BOSL2 #1683: Add support for varying mask_angle in a rounding mask

dan-p3rry, 2025-05-12:
**Is your feature request related to a problem? Please describe.**
I posted an item in Discussion today, wanting to apply fillet to a corner with varying mask_angle.  This is a specific solution for a 90 degree corner, but I think the code can be generalized to other angles.

**Describe the solution you'd like**
Add a fillet (rounding mask) to the circled corner to complete the fillet.  (The face facing left is not 90 degrees to the base.)

[image]

**Example Code**
In this code, changing cr to negative value gives an undercut fillet (and rotated 180 degrees)
```
include <BOSL2/std.scad>

$fs = 0.4; 
$fa = 2;

start_ma=80; 
end_ma=90;
fillet=3; 
cr=15;

// derived from skin() ex 16
shapes = [for(i=[0:$fa:90-$fa]) zrot(i, p=xrot(90, 
        p=right(cr, p=path3d(mask2d_roundover(r=fillet, 
            mask_angle=start_ma + (end_ma - start_ma)*(i/90))))))
    ];

skin(shapes, slices=0);

```

**Additional context**
Add any other context or screenshots about the feature request here.

adrianVmariano, 2025-05-13:
Does that align with the vertical rounding on the edge?  How do you envision this as a general capability?

dan-p3rry, 2025-05-13:
[image]

This is not an extreme angle (81 degrees), but yes, you can see good alignment to the rounded vertical edge.  

Regarding general capability, I think it should be a new module in masks3d.

adrianVmariano, 2025-05-13:
I'm curious about whether you can get the faceting of the fillet to match the faceting of the vertical edge.  The dark blue is kinda of hard to see, actually, but it does look good.  

Could this be made into a general capability that accepts any mask from mask2d?   That would give the user lots of flexibility.  There could be some basic cases supported directly.   But for example, you gave a fillet of constant radius, but when the angle varies, a fillet of constant height may look better.   On the other hand, maybe this idea isn't that useful if it assumes the vertical edges are circular roundings.  

Another idea would be to integrate this into prismoid().  

Why do you want to introduce path_sweep2d?  In particular, for this problem the curves are all straight segments with circular arcs, so you can compute the offsets directly.  Involving path_sweep2d, which calls offset() will be slow and will probably be less robust and will fail under some circumstance.

dan-p3rry, 2025-05-13:
[image]

In this image, you can see what happens when the mask_angle is < 90 degrees.  And you can see that this technique is not exact.  Any suggestions to improve the fit?

adrianVmariano, 2025-05-13:
Can you post the full code?  I find it hard to understand 3d models without being able to spin them around.   I think you can add it as a file if it's long and that's easier.  

My guess as to the problem you're having is that you have non 90 deg angles, which means some assumptions about fillet geometry may be failing.  In 90 deg, the width of a fillet or rounding is the same as the radius.  That's false at other angles.  

Another question occurred to me about API, which is this: how does the user specify the required angles of the fillet or rounding?  This is something may not easily known or easily computed.  This actually connects to a broader question of how to specify corner geometry and how corner anchors should work in order to be useful for something.

adrianVmariano, 2025-05-13:
Actually I think there's another complication.  The axis of rotation to align with the vertical edge needs to be the vertical edge.  But the axis of rotation to align with the base needs to be a line perpendicular to the base.   These are the same axis in the 90 deg case but not otherwise.  

Just fyi, the reason there's no support for this except rounded_prism() is that we didn't know how to do it.   Getting away from circular rounding (e.g. in rounded_prism()) makes it much easier.  

An attractive API concept would be to have edge_profile able to apply it.  I think it can get the angles of the edges.  But I think that's impossible because it takes geometry instead of a path.

dan-p3rry, 2025-05-13:
```
include <BOSL2/std.scad>

$fs = 0.4; $fa = 1;

size1 = [60, 10];
size2 = [50, 10];
h = 25;

prismoid_dan(size1=size1, size2=size2, h=h, rounding=3, roundingb=3);

module prismoid_dan(size1, size2, h, rounding, roundingb, roundingt) {
    base_angle_ = atan(h/(0.5*(size2.x - size1.x)));
    base_angle = base_angle_ > 0 ? base_angle_ : 180 + base_angle_;
    echo(base_angle=base_angle);
    // difference() {
        prismoid(size1, size2, h, rounding=rounding);

    // }

    if (roundingb > 0) { // add a fillet
            yrot(-90) fwd(0.5*size1.y) down(0.5*(size1.x - 2*rounding))
                zrot(-90)
                linear_extrude(size1.x - 2*rounding) 
                    mask2d_roundover(r=roundingb);
            yrot(-90) back(0.5*size1.y) down(0.5*(size1.x - 2*rounding))
                linear_extrude(size1.x - 2*rounding) 
                    mask2d_roundover(r=roundingb);

            left(0.5*size1.x) xrot(90) down(0.5*(size1.y - 2*rounding))
                mirror([1, 0, 0])
                linear_extrude(size1.y - 2*rounding) 
                mask2d_roundover(r=roundingb, mask_angle=base_angle);
            right(0.5*size1.x) xrot(90) down(0.5*(size1.y - 2*rounding))
                linear_extrude(size1.y - 2*rounding) 
                mask2d_roundover(r=roundingb, mask_angle=base_angle);

        // corners
        corner_sh = corner_round_variable_mask_angle(90, rounding,
                    roundingb, 90, base_angle);
            right(0.5*size1.x - rounding) back(0.5*(size1.y - 2*rounding))
                mirror([0, 1, 0]) zrot(270)
                    skin(corner_sh, slices=0);
            right(0.5*size1.x - rounding) fwd(0.5*(size1.y - 2*rounding))
                zrot(270)
                    skin(corner_sh, slices=0);

            left(0.5*size1.x - rounding) back(0.5*(size1.y - 2*rounding))
                zrot(90)
                    skin(corner_sh, slices=0);

            left(0.5*size1.x - rounding) fwd(0.5*(size1.y - 2*rounding))
                mirror([0, 1, 0]) zrot(90)
                    skin(corner_sh, slices=0);
    }
}

function corner_round_variable_mask_angle(angle, cr, fillet, start_ma, end_ma) =
    [for(i=[0:$fa:angle]) zrot(i, p=xrot(90, 
            p=right(cr, p=path3d(mask2d_roundover(r=fillet, 
            mask_angle=start_ma + (end_ma - start_ma)*(i/angle))))))];

```

dan-p3rry, 2025-05-13:
I agree with you that incorporating the code into prismoid is the better solution.  The above code can easily be modified to calculate four different mask_angles with unique rounding at each edge & corner.

I have an idea how to fix the errors at the corners, but it will take some time -- and lots of trig.

dan-p3rry, 2025-05-14:
I spent a full day on this, trying to develop a better solution to handle the corners, but didn't have success.  The imperfect solution above is as close as we're going to get without a LOT of work.

adrianVmariano, 2025-05-14:
Like I said, we couldn't figure out how to do it.   It may be better to seek a solution involving beziers or NURBS.  I don't know if it's possible to use the NURBS representation of a circle to create mating circular roundings---if you really care about circularity.   But I can tell you that the solution to handling corner roundings at arbitrary angles and with 2 different radii that meet at a corner was entirely straight forward with beziers using continuous curvature, as implemented in rounded_prism.   And more flexibility surely exists without the continuous curvature constraint, or with NURBS instead of beziers.

dan-p3rry, 2025-05-14:
rounded_prism() does what I need, thanks!

closed 2025-05-14 by dan-p3rry