BelfrySCAD/BOSL2 #1688: [BUG] regular_prism wrong std anchors when specifying chamfer

B14CK313, 2025-05-19:
`regular_prism` anchors in +X +Y direction are incorrect when using chamfer.

This occurs even for `chamfer = 0`, so I suspect specifying chamfer takes another code path that triggers the behaviour.

```openscad
include <BOSL2/std.scad> // acd46acde53ab710e67c5493789d6a2f6d12ded1

right(0) regular_prism(n = 8, id = 10, h = 3, realign = true) show_anchors(s = 3, custom = false);
right(15) regular_prism(n = 8, id = 10, h = 3, realign = true, chamfer = 0) show_anchors(s = 3, custom = false);
right(30) regular_prism(n = 8, id = 10, h = 3, realign = true, chamfer = 1) show_anchors(s = 3, custom = false);
```

[image]

Thanks for developing this great library!

adrianVmariano, 2025-05-20:
This is happening because there is a problem with computing anchors to VNFs, I think caused by triangulated faces or duplicated vertices.  It's going to require a complete rewrite of the relevant code section to fix this.

adrianVmariano, 2025-05-25:
I believe this is now fixed.  As noted, the fix involved a total rewrite of the VNF anchoring code.  Furthermore, it turns out that the angles of chamfers were incorrect, so the code for creating chamfers has been reworked.   So let me know if you notice any anomalies with the update.

closed 2025-05-25 by adrianVmariano

B14CK313, 2025-05-25:
Wow, thank you so much for the fast fix. Works great for me. The chamfer also looks consistent with other shapes now.
Everything works just fine for me, but if I notice any anomalies in the future, I will of course share them.

Many thanks for your work, much appreciated.