BelfrySCAD/BOSL2 #1697: [BUG] `openscad-docsgen` fails with  OpenSCAD version 2025.05.16.ai25460

ssteinerx, 2025-05-30:
**Describe the bug**
Generating the docs locally fails with three errors when using a newer OpenSCAD version than the one used in ` .github/workflows/gen_docs.yml`.
  
I have:
```bash
$ openscad --version
OpenSCAD version 2025.05.16.ai25460
```
`openscad-gocsgen` doesn't report a version -- it's a pull from a couple of days ago.

**Code To Reproduce Bug**
```bash
$  openscad-docsgen -f
```

**Expected behavior**
Generate docs, no errors.

**Actual behaviour**
There are three errors, and they're all the same WARNING that then leads to an assertion failure:
```bash
     WARNING: Viewall and autocenter disabled in favor of $vp*
```

```bash
  1691,1: !! ERROR at shapes3d.scad:149: Failed OpenSCAD script:
  1692:        Image: cuboid_10.png
  1693:        cmd-line = /usr/local/bin/openscad -o tmp_cuboid_10.png --imgsize=640,480 --view=axes,scales --colorscheme Cornfield --projection=o --preview  tmp_cuboid_10_png.scad
  1719,1: !! ERROR at shapes3d.scad:151: Failed OpenSCAD script:
  1720:        Image: cuboid_11.png
  1721:        cmd-line = /usr/local/bin/openscad -o tmp_cuboid_11.png --imgsize=640,480 --view=axes,scales --colorscheme Cornfield --projection=o --preview  tmp_cuboid_11_png.scad
  2364,1: !! ERROR at utility.scad:985: Failed OpenSCAD script:
  2365:        Image: shape_compare.png
  2366:        cmd-line = /usr/local/bin/openscad -o tmp_shape_compare.png --imgsize=640,480 --view=axes,scales --camera 0,0,0,55,0,25,444 --colorscheme Cornfield --projection=o --autocenter --viewall --preview  --hardwarnings tmp_shape_compare_png.scad
```

**Additional Info**
The only issue I could find about this $vp* in the OpenSCAD repo was about killing the warning when running in the GUI and it was from 2020.  I do not know whether this error also occurs in the earlier OpenSCAD releases but I didn't see it when I ran the github action locally with `nektos/act`.

adrianVmariano, 2025-06-02:
It appears that this may be the result of an OpenSCAD bug.   As far as I can tell, with recent dev versions you can't preview any more and CGAL has some kind of bug in it, so you have to use --backend=manifold.

ssteinerx, 2025-06-03:
It only happens where --viewall and --autocenter are used.

This bug: 

[let $vp.. variables reflect the viewport when rendering from the command line #4466](https://github.com/openscad/openscad/issues/4466)

This is the relevant quote:

> [jordanbrown0](https://github.com/jordanbrown0) Contributor
> [on Nov 29, 2023]
> Note that we cannot set the $vp* variables correctly for --autocenter or --viewall, because those don't know the camera 
> position until after the model is evaluated.

So the error message is that `WARNING: Viewall and autocenter disabled in favor of $vp*` is just saying that the --autocenter and --viewall parameters are disabled 'cause they can't work i.e. "viewall" and "autocenter" can't be done 'till you know what "all" and "center" mean by which time rendering is done. 

 It is saying that, if you want this, you have to use the $vp* variables in the script itself so it can, presumably, make these adjustments *while* rendering.

Why is it an error instead of a warning?  Dunno.

ssteinerx, 2025-06-03:
> Because cuboid is actually generating errors with recent snapshots.

I believe you, but this error message is pretty specific about the command line parameters, and I'm only talking about building the documentation.

So I'm sure that...
  - the documentation build fails
  - the error message I quoted is the one I got during the failed build
  - the error message explicitly references the `--autocenter` and `--viewall` parameters vis-à-vis the `$vp*` globals
  - the referenced issue and the developer's comment therein are directly related to these specific cmd line parameters

adrianVmariano, 2025-06-03:
I'm not sure how you reached the conclusion that the --autocenter and --viewall warning message was causing the failure.  Appearing during build doesn't mean it caused a failure.  The actual error messages you quote give no information at all about what failed.  Revar has indicated that those parameters are passed when no VPD, VPR, or VPT parameters are given for the example, and NOT passed if any of those parameters are given.  But the examples you list with failures actually **have** those viewport parameters.  Meanwhile, other examples without them don't appear with errors on your list.  This seems to be inconsistent with your interpretation.  

As noted, the error messages you quote give no information.  What happens if you go to one of the error lines, e.g. 149 of shapes3d.scad, extract the example, and run it yourself.  Do you get an error?  And if so, what is it?