BelfrySCAD/BOSL2 #1698: [BUG] affine* missing from documentation.

dan-p3rry, 2025-05-30:
From rounding.scad, rounded_prism() example 12: 

```
include <BOSL2/std.scad>
rounded_prism(path3d(pentagon(3)), apply(affine3d_skew_yz(0,-20),path3d(pentagon(3),3)),
              joint_top=0.5, joint_bot=0.5, joint_sides=0.5);
```

A search of the BOSL2 github area shows the function exists in affine.scad, but without documentation, it will be difficult to use.

I can get by with a xrot() command instead, just wanted to let you guys know ....

adrianVmariano, 2025-05-30:
Those are internal functions.  You're not supposed to use them.

adrianVmariano, 2025-05-30:
Note that you cannot get the same result with xrot().  I have fixed the example to use skew()

dan-p3rry, 2025-06-01:
This example would have gotten me to my result a lot faster (but perspective makes it difficult to see all three skews in one view):

[image]

```
include <BOSL2/std.scad>

xdistribute(spacing=15){
    skew(axy=15) square(10);
    skew(ayx=15) square(10);
    skew(azx=30) square(10);
}
```

dan-p3rry, 2025-06-01:
Yes, that example will help a lot.  azy and azx are unique in that they create a 3D shape from 2D shape, compared to axy/ayx.

closed 2025-06-01 by dan-p3rry

adrianVmariano, 2025-06-01:
Well, no, nothing creates a 3d shape from a 2d shape except extrusion.  Your use of "square" is invalid with azx.