BelfrySCAD/BOSL2 #1701: Thoughts on hooks

amatulic, 2025-05-31:
I've been thinking about a possible new file hook-n-ring.scad (or some similar name) that the code in #1687 could be merged into.

It occurs to me that there's a fairly large class of 2D wire hooks and 2D ribbon hooks, both of which are made by extruding a 2D profile upward, or sweeping a cross-section along a 2D path.

Here are some thoughts I sketched about the different common hook profiles. The hook itself could be on either end of the wire, and can be configured with a single bend or double bend (not counting the bend from centering the hook with respect to the "body"), and the bend can be enough to close the hook into a loop.

[image]

I have a "ribbon" hook on the door of a closet. The hook is just on one end, and a hole for fastening is on the other end. A bit thicker and plastic would work fine for this.

[image]

I've already written the code for the cross section of the "wire" or "ribbon". It respects $fn, $fa, and $fs. Radius of the rounding self-adjusts depending on the width and height, maintaining a 40° angle from the build plate.
````
include <BOSL2/std.scad>

function wire_hook_profile(width, height, overhang=40) =
    let(
        r = min(width/2, height/(2*cos(overhang))),
        curvht = r*cos(overhang),
        rxoff = max(0, width/2-r),
        ryoff = max(0, height/2-curvht),
        astep = $fn ? (358-4*overhang)/$fn : min($fa, (180/PI) * $fs/r)
    ) let(astop = 90-overhang) [
        for(a=[0:-astep:0.001-astop]) [r*cos(a)+rxoff, r*sin(a)-ryoff],
        [r*cos(-astop)+rxoff, r*sin(-astop)-ryoff],
        for(a=[180+astop:-astep:180.001]) [r*cos(a)-rxoff, r*sin(a)-ryoff],
        [-r-rxoff,-ryoff],
        for(a=[180:-astep:180.001-astop]) [r*cos(a)-rxoff, r*sin(a)+ryoff],
        [-r*cos(astop)-rxoff, r*sin(astop)+ryoff],
        for(a=[astop:-astep:0.001]) [r*cos(a)+rxoff, r*sin(a)+ryoff],
        [r+rxoff, ryoff]
];

polygon(points=wire_hook_profile(10, 5, $fa=5));
left(9) polygon(points=wire_hook_profile(5, 10, $fa=5));
right(11) polygon(points=wire_hook_profile(10, 10, $fa=5));
````
That code produces these three example profiles:

[image]

I was wondering what the API for wire hooks would look like. I envision two modules:
* One that takes dimensions for inner radius (outer not needed) of the bends, the length of the extension on the end of the hook, the bend direction, and the type of hook on each end.
* One that takes a path and applies rounding to each vertex as appropriate, possibly different for each edge.

amatulic, 2025-05-31:
There are many hook configurations. I just wanted to start out with a basic class of 2D hooks because they are versatile and easy to print. I thought a general-purpose module (or two) would cover most of the use cases for this class of hooks. They can be made into wall hooks, shelf hooks, coat hooks, door hooks, towel hooks, etc. [This hook library on Thingiverse](https://www.thingiverse.com/thing:1367661) is way more complicated than it needs to be, with many different dimensions defined (mostly to accommodate spacing between screws and features of the hook), as you can see by scrolling through the images. If you get a Google vignette, just hit your back button and go forward again.

Printables and Thingiverse have lots of different kinds of hooks, especially caribiners, which I think we should avoid unless we can come up with a superior design. I saw YouTube videos testing the strength of these, and they almost always fail at the latch, even when the latch is trapped with a collar. This surprised me because I expected the hinge to shear first, if the design included a hinge for the latch (some of them just flex).

dan-p3rry, 2025-05-31:
There's also a clothes-hanger style hook.  This is from a model I did 18 months ago to hang from a shower head arm.

[image]

adrianVmariano, 2025-05-31:
No, if you screw the hook to a wall, or mount it directly in your model (especially the latter because it enormously increases the mounting area) then strengthening the bottom of the hook will make a big difference.  

If you think a symmetric API makes sense that may suggest the helper-function with structures API approach as a better method.   Another advantage of that scheme is that it's pretty easy to expand it to include other geometries if someone comes up with them.  

As far as I can tell, @dan-p3rry's hook fits into the framework of the original post, being a circular hook that is centered.  Isn't that the case?