BelfrySCAD/BOSL2 #1707: [BUG]Combination of arc & stroke not giving 90 degree angle?

dan-p3rry, 2025-06-05:
**Describe the bug**
I used stroke() with arc() to form a bending 90 degree path.  The resulting shape didn't match up to the adjacent shapes because the angle is not 90 degrees,

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>

$fs = 0.4; 
$fa = 0.1;

linear_extrude(20, convexity=4) 
    stroke(arc(r=7.5, angle=90, start=180, cp=[7.5, 7.5]), 
                width=3, endcaps="butt");
```

(The unreasonable value of $fa is me trying to determine if the error is related to $fa, it's not.

**Expected behavior**
Flat edges at end of curve should be parallel to axes and 90 degrees apart.

**Screenshots**

[image]

**Additional context**
I was able to use path_sweep to avoid stroke(arc());

BOSL2 download date was 5/30/2025.  OpenSCAD version is 2025.05.06.

adrianVmariano, 2025-06-14:
Your first and last segments of the arc are not parallel to the axes, so this is the expected result.  You'll get the same behavior with path_sweep generally, unless you give user-defined tangents to correct control that last face.

dan-p3rry, 2025-06-14:
> You'll get the same behavior with path_sweep generally
With path_sweep(), the angle is 90.0045 degrees, considerably smaller error than stroke(arc()).

```
include <BOSL2/std.scad>

$fs = 0.4; 
$fa = 1;

linear_extrude(20, convexity=4) 
    stroke(arc(r=7.5, angle=90, start=180, cp=[7.5, 7.5]), 
                width=3, endcaps="butt");

path_sweep(rect([3, 20], anchor=BOTTOM+RIGHT), 
                        arc(r=7.5, angle=90, start=0, cp=[7.5, 7.5]));
```

The stroke() version and the path_sweep() version use the same arc() statement, so the strange behavior is coming from stroke().  I can make the error smaller by using unreasonably small values of $fa and $fs, or with an unreasonably large value of $fn in the arc() statement.  (My comment in the initial submission that $fa doesn't affect the error is incorrect, not sure what I did differently that time.)

My expectation is that the stroke(arc()) statement creates a lot of slices that form the curved shape, and that the very first slice would have one face parallel to the X-Z plane when start=180 -- which is true for path_sweep(arc()), not for stroke(arc()).

adrianVmariano, 2025-06-14:
No, the "strange behavior" is coming from path_sweep because path_sweep uses a multi-point derivative estimation to compute the tangent at each point on the path, whereas stroke simply gives you exactly what you feed it.  It creates the exact segments of your input path.  

Your expectation is simply wrong.  Examine the actual points of the arc.  You'll see that the first segment of the arc has the 93 deg angle or whatever.   Think about what it means to compute points on an arc: it is impossible for the first segment to have zero slope.  For that to occur you'd need to sample equidistantly on both sides of the axis.  Since stroke replaces every segment in the input with a rectangle of exactly the same position and length, you get out exactly what you put in.    This is fundamental: how could stroke produce a different result?  If you want to "fix" it you need a different arc that is actually zero slope at the ends.  Perhaps try a squircle?   Or use round_corners on a 3-segment path.

adrianVmariano, 2025-06-15:
Another way to get an arc with actual flat segments at the ends is to use a circumscribing realigned ellipse and then cut it:

```
arc = ellipse(10, circum=true, $fn=20, realign=true);
polygon(right_half(arc)[0]);
```

closed 2025-06-18 by adrianVmariano