BelfrySCAD/BOSL2 #1713: [BUG] text3d() does not have documented attachment capabilities

jstebbins, 2025-06-13:
**Describe the bug**
 The text3d documentation says, "You cannot attach children to text." 

I assumed this to mean I can attach text to a parent object. I.e.
```
cube(20) attach(FRONT, BOT) text3d("Test", h=1);
```
But I may not attach an object to text3d, e.g.
```
text3d("Test", h=1) attach(BOT, FRONT) cube(20);
```
But it seems neither are permitted. The former gives the assertion, "text3d() does not support parent-child anchor attachment with two parameters" and the latter gives the assertion, "Module text3d() does not support child modules".

If I comment out the assert in module text3d that causes the former warning, I get the expected results.  The orientation of the text changes as I change the 2nd parameter of `attach()`.

**Code To Reproduce Bug**
```
cube(20) attach(FRONT, BOT) text3d("Test", h=1);
```

**Expected behavior**
Expect to see a cube with text floating in front of it.

**Additional context**
OpenSCAD version 2025.03.26
BOSL2 git master

adrianVmariano, 2025-06-14:
This cannot be fixed due to limitations in OpenSCAD.  Evidently the documentation needs to be improved.  

You can do 

`cube(20) attach(FRONT) text3d("Test", h=1);`

using the single parameter form of attach().

jstebbins, 2025-06-14:
In my real code, text3d is invoked a couple levels down from where the actual `attach()` is happening. I.e. text3d has ancestor objects for which the `attach(FRONT, BACK)` is valid. I think the only way to reset '$attach_to' and keep the desired position and orientation is to do `position($attach_anchor[0]) text3d()`.

jstebbins, 2025-06-14:
Is the reason this can't work because you can't query the width of the text? So you can't set the required size for `attachable()`? If so, maybe it would be a better user experience to make it a full-fledged attachable, but set the width to 0. So 4 of 6 orientations will work as expected and you can easily document the mis-feature of anchoring to center for the other 2 I think.

adrianVmariano, 2025-06-14:
I'm closing this now.  You can reopen if you have any further thoughts.

closed 2025-06-14 by adrianVmariano