BelfrySCAD/BOSL2 #1741: Additional arguments for roof()

lar3ry, 2025-07-02:
**Is your feature request related to a problem? Please describe.**
I would like to embed text (and other things) into a surface and would like to use roof() to do it.
Currently, there is only one argument (voronoi), and the height is determined by the size of the objects parts, because roof() uses a 45 degree angle. 

**Describe the solution you'd like**
Provide the ability to specify either a numeric height or an angle to make the height uniform regardless of the sizes of the parts of the objects.
Additionally, allow negative numbers for the numbers to allow roof() on the bottom of the object

**Describe alternatives you've considered**
None, really. roof() looked like the best answer for the task.

**Example Code**
```
// Code goes here.
roof(d=30) text("HELLO");  // degrees = angle of roof.
roof(h=1) text("WORLD");  //  height = height of all parts of an object's roof.
// Prefixing a '-' to the numbers would make the roof on the bottom of the object.
```

**Additional context**
Add any other context or screenshots about the feature request here.

lar3ry, 2025-07-02:
Ahh! Should have known. It's such a nifty thing, I just assumed it was in my favourite library.

amatulic, 2025-07-03:
@revarbat - consider this: If the roof() algorithm could be implemented in BOSL2, the roof corner lines, when projected onto the plane of the original polygon, are equivalent to a [straight skeleton](https://doc.cgal.org/latest/Straight_skeleton_2/index.html), on which there has been a lot of literature written in the context of 2D offsets. It isn't a trivial problem.

The offset() function in BOSL2 works well enough for most cases, but I've experienced instances where it breaks down and produces nonsensical results.

Basically any 2D cross-section (or contour) of an object made with roof() is identical to a 2D negative offset of the original outline from which the roof was made.

It is an endless source of frustration to me that I have no access to the internal data of any OpenSCAD object. I have some interesting uses for reliable negative offsets.

amatulic, 2025-07-11:
I don't think there's an action for BOSL2 with regard to roof() unless you want to make a VNF roof from a straight skeleton, although if you can make a straight skeleton anyway, you already have what you need for a robust offset().

As far as offset(), would it be possible to have an option `fast=true`, which can be set to false to access the more robust version?

lar3ry, 2025-07-12:
Fine by me if it's closed. I'm happy with the solutions.

closed 2025-07-12 by lar3ry