BelfrySCAD/BOSL2 #1742: Allow quotation marks in text()

lar3ry, 2025-07-02:
**Is your feature request related to a problem? Please describe.**
There appears to be no way to include quotation marks in text.
While it is rare, I recently needed to use quotation marks in text to mean 'inches'.

**Describe the solution you'd like**
A way to include quotation marks in text()

**Describe alternatives you've considered**
Using ' to surround the quotation mark.
Using two ". 
Using an escape character in the same manner as the \ in Linux.

**Example Code**
```
// Code goes here.
```
text("4\" Nails");
text("4'"' Screws);
text("4"" Bolts");

**Additional context**
Add any other context or screenshots about the feature request here.

amatulic, 2025-07-03:
@revarbat - This issue should be closed, because there is nothing to fix, and has nothing to do with BOSL2 anyway. Quotation marks work perfectly fine in OpenSCAD. Just escape it with a backslash. This is already described in the OpenSCAD documentation; see https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/General#Strings

For example:
````
text("These are \"quotation marks\".");
````
Gives this:
[image]

lar3ry, 2025-07-03:
On Wed, 2025-07-02 at 14:23 -0700, Revar Desmera wrote:
> revarbat left a comment (BelfrySCAD/BOSL2#1742)
> I'm pretty sure OpenSCAD lets you escape a double-quote with a
> backslash: "4'5""

Yikes! I was sure I tried that. Yes, the backslash does the job.

> Also, if this were a bug, it'd be in OpenSCAD, not BOSL2.

Right! 

Thanks.