BelfrySCAD/BOSL2 #1759: [BUG] _slice_3dpolygons usage of polygon_area doesn't account for undef return

ferringb, 2025-07-27:
**Describe the bug**

Within _slice3dpolygons, it assumes that the return from polygon_area never returns undef (i[t can for nonplanar](https://github.com/BelfrySCAD/BOSL2/blob/d0bdc25da92dbeb1ddabb464429666926e66cf28/geometry.scad#L1506)), [thus does a direct comparison of the result against EPSILON](https://github.com/BelfrySCAD/BOSL2/blob/d0bdc25da92dbeb1ddabb464429666926e66cf28/vnf.scad#L1373)

This explodes with the expected `(undef>number)` complaint from openscad

**Code To Reproduce Bug**
The 3rd poly is the one that is nonplanar- `assert(is_undef(polygon_area([[0.8016, 0, 0], [1, 0.199601, 0.25], [0.7014, 0.5, 0.25], [0.5, 0.299401, 0]])))`.

This is the isolated call from some insane vnf crap I'm trying: 
```
polys =  [[[0.199601, 0, 0], [0.8016, 0, 0], [0.5, 0.299401, 0]], [[0.8016, 0, 0], [1, 0.199601, 0.25], [0.7014, 0.5, 0.25], [0.5, 0.299401, 0]], [[0.8016, 0, 0.25], [0.5, 0.299401, 0.25], [0.7014, 0.5, 0.5], [1, 0.199601, 0.5]], [[1, 0.199601, 0.5], [0.7014, 0.5, 0.5], [1, 0.8016, 0.5]], [[0.8016, 0, 0.25], [1, 0.199601, 0.5], [1, 0.199601, 0.25], [0.8016, 0, 0]], [[1, 0.8016, 0.5], [0.7014, 0.5, 0.5], [0.7014, 0.5, 0.25], [1, 0.8016, 0.25]], [[0.7014, 0.5, 0.5], [0.5, 0.299401, 0.25], [0.5, 0.299401, 0], [0.7014, 0.5, 0.25]], [[0.199601, 0, 0], [0.5, 0.299401, 0], [0.5, 0.299401, 0.25], [0.199601, 0, 0.25]], [[1, 0.199601, 0], [1, 0.8016, 0], [0.7014, 0.5, 0]], [[1, 0.8016, 0], [0.8016, 1, 0.25], [0.5, 0.7014, 0.25], [0.7014, 0.5, 0]], [[1, 0.8016, 0.25], [0.7014, 0.5, 0.25], [0.5, 0.7014, 0.5], [0.8016, 1, 0.5]], [[0.8016, 1, 0.5], [0.5, 0.7014, 0.5], [0.199601, 1, 0.5]], [[1, 0.8016, 0.25], [0.8016, 1, 0.5], [0.8016, 1, 0.25], [1, 0.8016, 0]], [[0.199601, 1, 0.5], [0.5, 0.7014, 0.5], [0.5, 0.7014, 0.25], [0.199601, 1, 0.25]], [[0.5, 0.7014, 0.5], [0.7014, 0.5, 0.25], [0.7014, 0.5, 0], [0.5, 0.7014, 0.25]], [[1, 0.199601, 0], [0.7014, 0.5, 0], [0.7014, 0.5, 0.25], [1, 0.199601, 0.25]], [[0.8016, 1, 0], [0.199601, 1, 0], [0.5, 0.7014, 0]], [[0.199601, 1, 0], [0, 0.8016, 0.25], [0.299401, 0.5, 0.25], [0.5, 0.7014, 0]], [[0.199601, 1, 0.25], [0.5, 0.7014, 0.25], [0.299401, 0.5, 0.5], [0, 0.8016, 0.5]], [[0, 0.8016, 0.5], [0.299401, 0.5, 0.5], [0, 0.199601, 0.5]], [[0.199601, 1, 0.25], [0, 0.8016, 0.5], [0, 0.8016, 0.25], [0.199601, 1, 0]], [[0, 0.199601, 0.5], [0.299401, 0.5, 0.5], [0.299401, 0.5, 0.25], [0, 0.199601, 0.25]], [[0.299401, 0.5, 0.5], [0.5, 0.7014, 0.25], [0.5, 0.7014, 0], [0.299401, 0.5, 0.25]], [[0.8016, 1, 0], [0.5, 0.7014, 0], [0.5, 0.7014, 0.25], [0.8016, 1, 0.25]], [[0, 0.8016, 0], [0, 0.199601, 0], [0.299401, 0.5, 0]], [[0, 0.199601, 0], [0.199601, 0, 0.25], [0.5, 0.299401, 0.25], [0.299401, 0.5, 0]], [[0, 0.199601, 0.25], [0.299401, 0.5, 0.25], [0.5, 0.299401, 0.5], [0.199601, 0, 0.5]], [[0.199601, 0, 0.5], [0.5, 0.299401, 0.5], [0.8016, 0, 0.5]], [[0, 0.199601, 0.25], [0.199601, 0, 0.5], [0.199601, 0, 0.25], [0, 0.199601, 0]], [[0.8016, 0, 0.5], [0.5, 0.299401, 0.5], [0.5, 0.299401, 0.25], [0.8016, 0, 0.25]], [[0.5, 0.299401, 0.5], [0.299401, 0.5, 0.25], [0.299401, 0.5, 0], [0.5, 0.299401, 0.25]], [[0, 0.8016, 0], [0.299401, 0.5, 0], [0.299401, 0.5, 0.25], [0, 0.8016, 0.25]]];
assert(
  _slice_3dpolygons(
    polys, "X", 
    [0.125, 0.25, 0.375, 0.5, 0.625, 0.75, 0.875]
));
```

This is the full inducing code.  Context, I'm playing around w/ the linear_extrude vnf code to generate a basket weave- my intention is to be able to make the actual 'hole' and 'width' of the weave controllable, thus the `width_gap` multiplier.  This is scratch code that doesn't produce what I want (even if I modify _slice_3d_polygons to ignore nonplanar).  I'm fine w/ my invocation being wrong- however asserts somewhere in the pathway in bosl2 should catch this before openscad tries comparing against an undef.

```
module weave(obj, weave_h, tex_size=[5,5], tex_depth=1, weave_gap=1) {
    assert(all_positive(tex_size) && is_list(tex_size) && len(tex_size) == 2);
    assert(is_num(weave_gap) && weave_gap > 0);
    // taken from https://github.com/BelfrySCAD/BOSL2/wiki/skin.scad#functionmodule-linear_sweep
    diag_weave_vnf = [
        [ //points
            [0.2, 0, 0], [0.8, 0, 0], [1, 0.2, 0.5], [1, 0.8, 0.5], [0.7, 0.5, 0.5],
            [0.5, 0.3, 0], [0.2, 0, 0.5], [0.8, 0, 0.5], [1, 0.2, 1], [1, 0.8, 1],
            [0.7, 0.5, 1], [0.5, 0.3, 0.5], [1, 0.2, 0], [1, 0.8, 0], [0.8, 1, 0.5],
            [0.2, 1, 0.5], [0.5, 0.7, 0.5], [0.7, 0.5, 0], [0.8, 1, 1], [0.2, 1, 1],
            [0.5, 0.7, 1], [0.8, 1, 0], [0.2, 1, 0], [0, 0.8, 0.5], [0, 0.2, 0.5],
            [0.3, 0.5, 0.5], [0.5, 0.7, 0], [0, 0.8, 1], [0, 0.2, 1], [0.3, 0.5, 1],
            [0, 0.8, 0], [0, 0.2, 0], [0.3, 0.5, 0], [0.2, 0, 1], [0.8, 0, 1], [0.5, 0.3, 1]
        ],[ //faces
            [0, 1, 5], [1, 2, 4, 5], [7, 11, 10, 8], [8, 10, 9], [7, 8, 2, 1], [9, 10, 4, 3],
            [10, 11, 5, 4], [0, 5, 11, 6], [12, 13, 17], [13, 14, 16, 17], [3, 4, 20, 18],
            [18, 20, 19], [3, 18, 14, 13], [19, 20, 16, 15], [20, 4, 17, 16], [12, 17, 4, 2],
            [21, 22, 26], [22, 23, 25, 26], [15, 16, 29, 27], [27, 29, 28], [15, 27, 23, 22],
            [28, 29, 25, 24], [29, 16, 26, 25], [21, 26, 16, 14], [30, 31, 32], [31, 6, 11, 32],
            [24, 25, 35, 33], [33, 35, 34], [24, 33, 6, 31], [34, 35, 11, 7],
            [35, 25, 32, 11], [30, 32, 25, 23]
        ]
    ];
    f = function(val) (val == 1 || val == 0 || val == .5) ? val : (val > 0.5 ? val * weave_gap : val / weave_gap);
    tex=[
        // points
        [for (points=diag_weave_vnf[0])
            [f(points.x), f(points.y), points.z * tex_depth]
        ],
        //faces
        diag_weave_vnf[1]
    ];
    for (x=tex[0]) echo(x);
    // added purely to demonstrate bosl2 considers this valid.
    vnf_validate(tex);

    linear_sweep(
        obj,
        h=weave_h,
        texture=tex,
        tex_size=tex_size,
        convexity=12);
};
weave(square(30), 10, weave_gap=1.02);
```

**Expected behavior**
This should protect against nonplanar inputs resulting in `undef`.

I've no clue if this function should assert against them, and if anything from `linear_extrude` down should've caught the vnf and rejected it.

I've no clue if this function should actually accept nonplanar, but the vnf fed into linear_extrude *is* considered valid by bosl2.

**Additional context**
BOSL2 version: this [commit](https://github.com/BelfrySCAD/BOSL2/commit/d0bdc25da92dbeb1ddabb464429666926e66cf28).
openscad version: 2005.05.01

amatulic, 2025-07-27:
Interesting. I created the original basic diagonal basket weave shown in the examples, and I had been intending to do something similar to your project: make the hole size, ribbon thickness, and ribbon width adjustable, and make the ribbons have rounded bends instead of sharp bends. I hadn't gotten to it yet.

I was about to reply that _slice_3dpolygons is an internal function not meant to be called as a user function, and as such it doesn't do checks for things that it assumes would already be OK (like it's assuming the input is coplanar). Then I see in your weave example that you aren't actually calling it directly.

One thing you could try is pass all polygons through `vnf_triangulate()` to get a list of triangle polygons, all of which are guaranteed to be planar.

What solution would you propose? Assert an error if it sees a nonplanar polygon, or somehow slice of the nonplanar polygon, which would likely require it to be triangulated first?

amatulic, 2025-07-27:
Bear in mind that my original weave was mostly hand-made, and I intended to make the improved parametric one also by hand. Basically I made one diagonal ribbon and then made 3 rotations of it. I do recall using BOSL2 on the final shape to remove duplicate points and redundant triangles, though, simplifying it to the point where it was a reasonable size when presented as raw arrays of points and faces in the examples.

adrianVmariano, 2025-07-28:
There is an assert 3 lines later that bails if the input is not coplanar.  Apparently I hadn't realized that the earlier calculation could fail.   I'm not sure why vnf_validate doesn't catch the non-coplanar face.

amatulic, 2025-07-29:
[According to the wiki](https://github.com/BelfrySCAD/BOSL2/wiki/vnf.scad#module-vnf_validate), `vnf_validate()` does check for non-planar faces. However, a valid VNF can have non-planar faces and still be manifold. If you pass polyhedron data into OpenSCAD's `polyhedron()` with non-planar faces, it works fine and OpenSCAD triangulates them automatically.

Asserting an error would require checking every face for planarity. I'm wondering if it would be better to pass the VNF into `vnf_triangulate()` instead, which would make every face planar.

ferringb, 2025-07-29:
@amatulic ignoring my bug report here, would bosl2 be open to a weave texture?  I've not yet got 'clean' code I'd ask people to review, but I've got it working sans some defects I need to find cycles to fix.  I'm using the path approach I mentioned so it should allow folks to do things like adjust the gap, or angles- albeit I'm going to have to do a N+1 for the angle thing.

Either way, if bosl2 is open to adding the texture, I'll open a ticket when I have some non-shit code I think is ready for consideration (soon).

adrianVmariano, 2025-07-29:
My only question about adding a weave texture is: how many different ways are they to create a weave texture?  I'd prefer to avoid "weave1", "weave2", "weave3" and so on, which could happen if we realize after adding one texture that there's a second great way to do it.   It appears that if you can adjust the wave angle, gap and width that is pretty general, though.  But @amatulic may have deeper insight since he's worked on this before.

ferringb, 2025-07-29:
@adrianVmariano yeah, I had that thought.  I think the existing function/modules taking `texture=string` should just take `texture="weave"`- a default.  One that is constrained in it's "general" shape, but can be adjusted internally for VNF refinement or to make it a tad bit less ugly.  As to implementation: that `texture="weave"` just internally sources the vnf from `create_weave_texture` and passes defaults in.

For folks doing what I'm after- we just invoke `create_weave_texture` and provide the relevant arguments, then pass that texture into stuff like linear_sweep.  It also ensure you don't have to dirty up all the various APIs that take textures.

Either way- I know the code I have can do gaps.  Angle is the next thing I want to tackle, and more to the point, I want to allow refinement of the path.  If you check the code I dumped above, it's basically hardcoded transitions- that can be made into a curve.   If you check the models in https://www.printables.com/model/1268555-lock-in-place-in-spool-desiccant-silica-gel-contai/comments/2762763 (and my comment), I've got code doing both- I need to iterate, but I think making the angle configurable (and some tweaks) would allow creating both patterns from a common implementation.  Emphasis on "I think".

Being we're talking it out- best we cut a seperate issue to track this?

adrianVmariano, 2025-07-30:
Yeah, separate issue probably a good idea.  Note that all the internal textures are accessed via the texture() function, so any new textures would similarly be accessed.  But you still have to name it.  And if there are three kinds of weaves we want to know that up front so the names are good.  That was my concern.

adrianVmariano, 2025-07-30:
Fix for this was merged.

closed 2025-07-30 by adrianVmariano