BelfrySCAD/BOSL2 #1763: Create diagonal flip_copy options. e.g. xyflip_copy

j-weck, 2025-07-29:
I have a structure, which consists of a parent with two children. The two children are the same, just at different positions. So I would like to flip_copy them. But the position has to be at the DIAGONAL other end of the parent. So I need to flip_copy it aorund the y AND the x axis. Currently, I can use either the yflip_copy or the xflip_copy, but not both. 

It would be nice to hava three more flip_copy function:
+ xyflip_copy
+ xzflip_copy
+ yzflip_copy

in that way you can copy children also to the diagonal end of a parent.

example:
```
include <BOSL2/std.scad>

cuboid([10,40,2]){
    color([0.255, 0.412, 0.882]) position(FRONT) cuboid([5,5,2], anchor = LEFT+BACK);
    color([0.255, 0.412, 0.882]) position(BACK) cuboid([5,5,2], anchor = RIGHT+FRONT);
}
```

[image]

j-weck, 2025-08-08:
Thanks for the response! That was also my approach. But the result is not what I need (compare the following result with the desired arrangement above):

```
include <BOSL2/std.scad>

cuboid([10,40,2]){
    //color([0.255, 0.412, 0.882]) position(FRONT) cuboid([5,5,2], anchor = LEFT+BACK);
    mirror_copy([1,1,0])
    color([0.255, 0.412, 0.882]) position(BACK) cuboid([5,5,2], anchor = RIGHT+FRONT);
}
```

[image]

I can't find a combination for mirror_copy, where I get the desired result :/

j-weck, 2025-08-08:
I found a solution. Not as nice as mirror_copy, but it works:
I seperate the children object in a module and add it two times to the parent object.
One time I rotate the module 180 degree around the z axis.

```
include <BOSL2/std.scad>

module small_cube(){
  color([0.255, 0.412, 0.882]) position(BACK) cuboid([5,5,2], anchor = RIGHT+FRONT);
}

cuboid([10,40,2]){
  small_cube();
  zrot(180) small_cube();    
}
```

[image]

adrianVmariano, 2025-08-08:
Rotate_copy exists.  

```
cuboid([10,40,2]){
    zrot_copies(n=2)
    color([0.255, 0.412, 0.882]) position(BACK) cuboid([5,5,2], anchor = RIGHT+FRONT);
}

closed 2025-08-08 by adrianVmariano