BelfrySCAD/BOSL2 #1774: cleanup: many unused definitions in threading.scad

coryrc, 2025-08-17:
* https://github.com/BelfrySCAD/BOSL2/blob/d252f781232a9262583806fe05bbb8529c108899/threading.scad#L1255
* https://github.com/BelfrySCAD/BOSL2/blob/d252f781232a9262583806fe05bbb8529c108899/threading.scad#L253
* https://github.com/BelfrySCAD/BOSL2/blob/d252f781232a9262583806fe05bbb8529c108899/threading.scad#L376

I've only run across these randomly and have not thoroughly searched for more.

Quick Google search, maybe https://pypi.org/project/sca2d can find unused variables? It also looks actively-worked-on so it may soon if it doesn't already.

adrianVmariano, 2025-08-21:
If you want to submit a PR that removes the many oprofile definitions and the one reference to depth that's not used I'll merge it.

coryrc, 2025-08-22:
Thanks, if nobody else gets to it, I will in the next few weeks.