BelfrySCAD/BOSL2 #1777: Non-manifold edges out of path_sweep()

dan-p3rry, 2025-08-20:
[https://github.com/openscad/openscad/issues/6114](url)

CC'ing FYI.  Since Openscad is rendering without error, I don't think it's an issue with path_sweep().  On the other hand, I've seen the issue multiple times with path_sweep().

adrianVmariano, 2025-08-20:
Major bug in path_sweep here:  it's adding endcaps even when closed=true.

dan-p3rry, 2025-08-20:
> Major bug in path_sweep here: it's adding endcaps even when closed=true. You can workaround by explicitly setting caps=false.

Confirmed that caps=false eliminates the non-manifold edges.  But this doesn't resolve the ticket at Openscad -- OpenSCAD should never output a non-manifold mesh.

closed 2025-08-21 by adrianVmariano

adrianVmariano, 2025-08-21:
No, as soon as the PR is merged it's live on github and when you download you'll get that version.  If you're seeing issues relating to the spurious caps still then reopen this and post your sample code.  It is easy to create invalid polyhedra with path_sweep even when everything is working right, though, so confirm maybe with vnf_validate that you didn't create self-intersection some other way.