BelfrySCAD/BOSL2 #1790: add Internal=true in get_profile_shift()

andrestu, 2025-09-01:
I would like to be able to get the right profile shift needed to achieve a desired spacing between a ring_gear and a mated spur_gear.

for example:
pshift = get_profile_shift(desired, teeth1, teeth2, mod=mod, internal=true );

or is possible to obtain this value from the current get_profile_shift() implementation or other calculation?

adrianVmariano, 2025-09-01:
You can currently get the required profile shift by giving get_profile shift a negative tooth value for the spur gear in the gear pair.  

However, the resulting profile shift has to equal the **difference** of the profile shifts of the gear pair, namely `ps_ring - ps_spur`.  Unless you have a lot of teeth on your spur gear and can accept a negative profile shift there, you're going to be stuck putting all the profile shift on the ring gear, so the adjustability is kind of limited.   Note that you can never decrease the distance between them, only increase it, since `ps_ring >= ps_spur` is required.  

Do you think it's worth adding options to support this more clearly in get_profile_shift given the complexity?

adrianVmariano, 2025-09-02:
I went ahead and added internal options to get_profile_shift.

closed 2025-09-02 by adrianVmariano