BelfrySCAD/BOSL2 #1795: Converting local vectors to world coordinates

kana-lab, 2025-09-06:
Hi,

# Context

I encountered a situation where I need to check whether an object lies inside a certain space.
For example:
```scad
cube([50, 50, 50], center=true) {
    attach(TOP, BOT)
    cube([30, 30, 30]);  // Does this lie inside a sphere(r=90)?
}
```

My idea to solve this problem is like this:
```scad
space_limit = 70;

cube([50, 50, 50], center=true) {
    attach(TOP, BOT)
    cube([30, 30, 30]) {
        local_vertices = vnf_vertices(cube([30, 30, 30], anchor=CENTER));
        global_vertices = some_local_to_global_transform(local_vertices, ...);
        farthest_idx = furthest_point([0, 0, 0], global_vertices);
        assert(norm(global_vertices[farthest_idx]) < space_limit, "check failed!");
    }
}
```

# My Questions

1) I couldn’t find any function to convert local vectors to global coordinates. Is there one?
2) If not, would it make sense to add such a function?
3) Is there a more proper or recommended way to solve this problem?

<details><summary>Thoughts about adding such a function</summary>

It seems straightforward to implement this conversion, since simply multiplying `$transform` by the local vector does the job.

Example:
```scad
function world_coords(p) =
    apply($transform, p);
````

However, when `p` is a long list of vectors, I observed that this sometimes returns unexpected results.
I suspect this may be because the evaluation order is not guaranteed to be the same as the code order,
similar to why `parent()` requires being wrapped in a `let` in OpenSCAD 2021.01.

Adding a dummy dependency might address this issue:
```scad
function world_coords(p, dummy_desc) =
    apply($transform, p);

PARENT() let(desc = parent()) {
    local_v = get_vertices(...);
    global_v = world_coords(p=local_v, dummy_desc=desc);
}
```

If this behavior is indeed due to OpenSCAD evaluation order, it might be worth documenting.
</details>

Thank you in advance!

adrianVmariano, 2025-09-06:
Generally the kind of problem you're trying to solve is impossible because OpenSCAD doesn't provide access to geometry data, so you can't know how big any particular object is.  But if you just want to know the relation between a point in one coordinate system to the global system, that is possible using descriptions.  

https://github.com/BelfrySCAD/BOSL2/wiki/attachments.scad#section-attachable-descriptions-for-operating-on-attachables-or-restoring-a-previous-state

See if you can solve your problem with that.

kana-lab, 2025-09-07:
Oh I didn't notice descriptions work effectively on this case.
Yeah I only need the distance between a local point and the origin and there's some relativity.

At first glance `desc_point` transforms the global coordinates to local ones, which is the inverse direction of transformation I need, so I thought it's not applicable to my case.
However we can still obtain the distance by transforming the origin to the local coordinates like:
```scad
origin_in_local_system = desc_point(desc, p=[0, 0, 0]);
farthest_idx = furthest_point(origin_in_local_system, verts);
longest_dist = norm(verts[farthest_idx] - origin_in_local_system);
```
You mean this usage right?
Thank you for your suggestion.

Btw how about adding the functionality to calculate the global coordinates?
It's not impossible as I mentioned above (in \<details> section).
I feel transforming local vectors to global system is more intuitive in this case.

kana-lab, 2025-09-07:
Thanks for the additional hints!

Maybe I misunderstood, but `desc_dist()` seems a bit limited for my use case since it only accepts anchors rather than arbitrary vectors.

Also, I might be wrong, but when you say `global=parent()`, doesn't that only allow global-to-local transformations?  
I'm still not sure how to perform local-to-global transformation directly without something like `apply($transform, vec)`.

Am I missing something here?

Anyway, your explanation gave me a much clearer picture — thank you again!

closed 2025-09-08 by kana-lab